magazino/pylon_camera #1: Error when testing a pylon dart

beatrizleon, 2016-06-15:
I get the following error when launching `roslaunch pylon_camera pylon_camera_node.launch`:

[image]

It seems that only mono8 cameras are supported. Is that true or I am missing something?

marceldebout, 2016-06-15:
> image <https://cloud.githubusercontent.com/assets/12493239/16074377/1de80714-32e2-11e6-925e-875f48d0425b.png>

Hello Beatriz,

you're alright - up to now the package only supports MONO8 cameras.
If a camera supports color encodings as well as MONO8, the latter will be selected.
Unfortunately the daA1280-54uc you are using does not provide MONO8 images.

http://www.baslerweb.com/en/products/cameras/area-scan-cameras/dart/daa1280-54uc

Best regards,
Marcel

beatrizleon, 2016-06-15:
Hi Marcel, 

Thanks for your answer. 

Please let me know when you add support for color images. We created a package that is generating ros images but I am having problems with the color format (between bgr and rgb). I am really interested in testing anorther solution to see if that manage to solve that problem.

Thanks,
Beatriz

closed 2016-06-15 by magazino