magazino/pylon_camera #3: "startGrabbing: Node is not writable." in pylon_camera package

andermi, 2016-06-21:
When I try to launch the pylon_camera node, I receive the following error:

> startGrabbing: Node is not writable.

I traced through the code and found that an exception is thrown when the node is trying to set the PixelFormat to Mono8 using the setValue function in pylon_camera_base.hpp in the startGrabbing function.

I'm using the basler ace acA4600-7gc GigE camera with pylon 5 sdk. I can view the camera images and set camera parameters just fine using the PylonViewerApp.

[http://answers.ros.org/question/237591/startgrabbing-node-is-not-writable-in-pylon_camera-package/](url)

marceldebout, 2016-06-22:
Thanks for reporting! I could reproduce the error using a acA4600-10uc and will try to provide a fix within the next days.

andermi, 2016-06-22:
Great, thank you! If you need any more info from me, please let me know!

marceldebout, 2016-06-23:
Done! Please leave a feedback if your problem could be solved.

andermi, 2016-06-23:
That seems to have done the trick! Thank you for such a quick response!

closed 2016-06-27 by andermi