magazino/pylon_camera #15: Hardware trigger mode

irazr, 2017-03-08:
Hi, does this package support hardware trigger mode?

marceldebout, 2017-03-08:
No, this pkg does not support the hardware trigger mode.
It's furthermore not planned to be implemented in the future.

Best,
Marcel

magazino, 2017-03-08:
Feel free to add it, in  case you need it ;-)

irazr, 2017-03-08:
Yep thanks, I'll have a try, because in my case, two cameras should be opened at the same time.

Best,
Yang

mgrrx, 2017-10-13:
@irazr do you have updates on this? Your contribution would be more than welcome!

Philpax, 2018-03-31:
I'd also like to check if anyone's had any luck with adding hardware trigger functionality; we also need it for stereo camera use.

AJwgnr, 2019-05-23:
Just found this. So I need the hardware trigger for a custom stereo vision system and would anyhow implement it.

I would write it that way, that the user can activate the hardware trigger in the  **config/default.yaml** file.
Have there been any discussion or advices that have been discussed before regarding this implementation? Otherwise I would submit my code as a pull request the next days.

yasmeenel3sh, 2019-05-30:
@AJwgnr  can you give us an update when you do it ,if you will ,I need it as well.

yasmeenel3sh, 2019-06-04:
@AJwgnr  ,I implemented it the way you said in this fork ,but I commented out the support for usb cameras because I needed to use ptp as well , https://github.com/yasmeenel3sh/pylon_camera/blob/indigo-devel/include/pylon_camera/internal/impl/pylon_camera_base.hpp