magazino/pylon_camera #16: Feature request: Automatic configuration in case multiple gige cameras are connected

marceldebout, 2017-03-09:
This affects the applyCamSpecificStartupSettings() method in the pylon_camera_gige.hpp.
`
cam_->GevSCPSPacketSize.SetValue(parameters.mtu_size_);                                                                                                                        cam_->GevSCPD.SetValue(parameters.inter_pkg_delay_); 
`
The inter-pkg-delay should be adapted but one has to figure out how this can be done. As far as I know, this depends the image size in bytes and the number of connected cameras. The mtu size might no longer be a required parameter, but can be read out of the network config instead. The number of connected cameras should be a parameter instead.
Further documentation can be found [here](http://s.baslerweb.com/media/documents/AW00064902000%20Control%20Packet%20Timing%20With%20Delays.pdf)

jonbinney, 2017-04-21:
@marceldebout I get a 404 error when I follow the documentation link you posted.

marceldebout, 2017-04-24:
@jonbinney [This is the desired document](https://github.com/magazino/pylon_camera/files/951740/AW00064902000_Controlling_Packet_Timing_With_Delays.pdf)

But I think this issue can be seen as closed, because the inter-pkg-delay is already provided as a parameter and and can't be calculated directly. It's furthermore recommended to be figured out via trial and error with the usage of the PylonViewerApp. The number of lost frames depends the cabling, the NIC as well as the amount of launched ros-nodes (the communication via the network stack). By increasing the inter-pkg-delay lost frames can be prevented by taking into account a lower maximum framerate. Also the number of cameras triggered in parallel can be configured by adapting the inter-pkg-delay parameter.

closed 2017-04-27 by marceldebout