magazino/pylon_camera #17: Feature request: Support 12bit pixel depth

marceldebout, 2017-03-09:
Currently only 8bit pixel depth is supported. The brightness / exposure search and different other functions are operating on values between 0 and 255. The support of 12bit pixel depth would need to template multiple functions that take `uint8_t` as default pixel type by taking into account the number of channels.

Nguyen-Quang, 2018-11-19:
Are you still working on it @marceldebout? I really want to take advantages of wide dynamic range camera.
Which are the parts of the codes that I need to template to make it support 12bit pixel?

Nguyen-Quang, 2018-11-19:
Are you still working on it @marceldebout? I really want to take advantages of wide dynamic range cameras.
Which are the parts of the codes that I need to template to make it support 12bit pixel?

Nguyen-Quang, 2018-11-19:
Are you still working on it @marceldebout? I really want to take advantages of wide dynamic range cameras.
Which are the parts of the codes that I need to template to make it supports12bit pixel?

Nguyen-Quang, 2018-11-19:
Are you still working on it @marceldebout? I really want to take advantages of wide dynamic range cameras.
Which are the parts of the codes that I need to template to make it supports 12bit pixel?

lucasw, 2018-11-20:
Crosslink https://answers.ros.org/question/308885/capturing-12-bit-with-pylon_camera/