magazino/pylon_camera #22: Pylon library download URL has changed

jonbinney, 2017-05-03:
At least it appears that way. The rosdep rule to install the pylon library just started failing for me with:

```
--2017-05-02 16:37:31--  http://s.baslerweb.com/media/documents/pylon-5.0.1.6388-x86_64.tar.gz
Resolving s.baslerweb.com (s.baslerweb.com)... 8.253.43.254, 8.254.246.254, 8.27.253.125
Connecting to s.baslerweb.com (s.baslerweb.com)|8.253.43.254|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://www.baslerweb.com/media/documents/pylon-5.0.1.6388-x86_64.tar.gz [following]
--2017-05-02 16:37:31--  https://www.baslerweb.com/media/documents/pylon-5.0.1.6388-x86_64.tar.gz
Resolving www.baslerweb.com (www.baslerweb.com)... 174.35.25.26, 174.35.25.24
Connecting to www.baslerweb.com (www.baslerweb.com)|174.35.25.26|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://www.baslerweb.com/en/redirect.php?request=26e1b8d2c087b3848b57ed845628fe5b [following]
--2017-05-02 16:37:31--  https://www.baslerweb.com/en/redirect.php?request=26e1b8d2c087b3848b57ed845628fe5b
Reusing existing connection to www.baslerweb.com:443.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: /en/support/downloads/document-downloads/ [following]
--2017-05-02 16:37:32--  https://www.baslerweb.com/en/support/downloads/document-downloads/
Reusing existing connection to www.baslerweb.com:443.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: ‘pylon-5.0.1.6388-x86_64.tar.gz’

pylon-5.0.1.6388-x86_64.tar.gz                         [ <=>                                                                                                             ] 286.26K  1.80MB/s    in 0.2s    

2017-05-02 16:37:32 (1.80 MB/s) - ‘pylon-5.0.1.6388-x86_64.tar.gz’ saved [293130]

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
```

The URL for downloading the package redirects to https://www.baslerweb.com/en/support/downloads/document-downloads/, which is a general page for downloading various documents. I think that the correct URL should be https://www.baslerweb.com/media/downloads/software/pylon_software/<filename>

I've updated the rosdep rule in our fork, and it now works: https://github.com/iron-ox/pylon_camera/commit/c49e0762ef6b99b128fa26d310214bd6c34e8459#diff-09bbd2e96d45b3d48a2f272e99cf0d6c

Only the change in first file there needs to be merged in to this repo. I had to add the second so I could test. I'll make a PR with the change, but I'd also like to know that other people are running into the same issue, and this isn't just some temporary problem with basler's servers.

simonbogh, 2017-05-15:
I have the exact same issue, and you are correct, the URL has changed.

jonbinney, 2017-06-02:
Hmm... the new URL is now failing.... looks like they may have moved it again. Investigating.

jonbinney, 2017-06-02:
Ugh they named the file for the 5.0.1 version `pylon-5016388-x86_64tar.gz` (note that there is no `.` before `.tar`!). Also no `.` between the version numbers.

jonbinney, 2017-06-02:
I've hacked this PR to work. The tarball name changed, and now the tarball name is different than the name of the directory it contains. If anybody knows anyone at basler, curse at them for me...

NikolasE, 2017-08-19:
Basler finally also published a dpkg-package (that also includes Qt) so the installation progress is probably becoming easier in the future.

commit b70bf81a (2018-02-13): Installation via pylon-debian pkg

This fixes #22, fixes #31
closed 2018-02-13 by MagazinoJenkins