magazino/pylon_camera #25: Increase image frame rate

zhangyongwei93, 2017-08-17:
Hello,
The camera I used is basler daA1600-60uc, USB interface,  ROS is kinetic, when I use pylon viewer software to capture image with **Bayer8** format, the frame rate can up to 60fps, but when I use this ROS package with **bayer_rggb8** format , the output include this 
<pre><code>
Desired framerate 60.00 is higher than max possible. Will limit framerate to: 30.60 Hz
</code></pre>
the image can capture normally， but the image topic only 25.4fps. How to increase the frame rate?
Thanks

NikolasE, 2017-08-17:
What's your exposure?

NikolasE, 2017-08-19:
What's your exposure when you activate exposure_auto? @marceldebout

qinvav, 2019-05-14:
Hello. I’m experiencing the same issue. acA2040-35gc, Bayer 2048x1536, Ethernet, ROS Kinetic.
1. Default settings, resulting picture is way to dark.
	Pylon Viewer – 34fps, exposure(us) – 3000;
	ROS driver – 28fps, , exposure(us) – 3000.
2. ”Auto Functions” settings, resulting picture is very bright.
	Pylon Viewer – 34fps, exposure(us) – 27910;
	ROS driver – 16fps, , exposure(us) – 19299.
3. In general: longer exposure – fewer fps, as if the exposure time adds to frame period.

Do you have any ideas what could cause this?

SherifN, 2019-06-21:
I am also having the same issue with daA1600-60uc 1600x1200, USB interface, ROS melodic. 
Pylon Viewer ~41fps
ROS Node ~ 20fps 
Is there already a solution?

kspons, 2019-12-05:
same issue with acA2500-60uc. in ROS will not go above 18.84 Hz

qinvav, 2019-12-06:
I might be wrong but I believe that the reason for the difference in FPS is different grabbing strategies. Pylon viewer uses so-called “free-running” mode, which yields maximum FPS. This driver uses (presumably) WaitForFrameTriggerReady() + ExecuteSoftwareTrigger(), which is slower, but necessary for stereo/multiple cameras synchronization (I guess). If someone is not afraid to get his hands a bit dirty I could share my driver for GigE cameras (should work for other types with minor modifications), it is fast :-)

claudiofantacci, 2020-03-06:
May be related to https://github.com/basler/pylon-ros-camera/issues/28 and/or https://github.com/basler/pylon-ros-camera/issues/29, even thought the implementation may be different in some part.