magazino/pylon_camera #26: Binning and brightness bug

xepost, 2017-09-08:
Hi,

I think I find a  minor bug. I recently got some pylon cameras, when I was playing  with the package I tried to use  binning and set the brightness to a  set value but it does not  seem to work.  Ii get the following error message.

[ WARN] [1504852077.113283476]: [] name not valid for camera_info_manger
[ INFO] [1504852077.141460250]: CameraInfoURL needed for rectification! ROS-Param: '/pylon_camera_node/camera_info_url' = '' is invalid!
[ WARN] [1504852077.141528307]: Will only provide distorted /image_raw images!
[ INFO] [1504852077.172282456]: Setting horizontal binning_x to 2
[ WARN] [1504852077.172356193]: The image width of the camera_info-msg will be adapted, so that the binning_x value in
[ INFO] [1504852077.200125703]: Setting vertical binning_y to 2
[ WARN] [1504852077.200154708]: The image height of the camera_info-msg will be adapted, so that the binning_y value i
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 576240) >= this->size() (which is 576000)
[pylon_camera_node-2] process has died [pid 3716, exit code -6, cmd /home/xepost/catkin_ws/devel/lib/pylon_camera/pylolon_camera_node __log:=/home/xepost/.ros/log/d93f9a52-945e-11e7-be19-80fa5b351634/pylon_camera_node-2.log].
log file: /home/xepost/.ros/log/d93f9a52-945e-11e7-be19-80fa5b351634/pylon_camera_node-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing

I change the  binning values to binning_x: 1  binning_y: 1 and it starts to work again.   Any idea how to fix it?

Thanks!

marceldebout, 2017-09-08:
Thanks for reporting! You're right this is a logical bug which we didn't think about up to now. The mean brightness of the image is calculated by only taking into account a subset of pixel. This subset is represented as a vector of indices. When changing the image size, this index vector is not recalculated and the access leads to above vector-range-error. I'll think about a solution and will push a fix next week.

xepost, 2017-09-08:
Hi Marcel,

If  I can come up  with some other  solution before  you do I will let you know, but seems unlikely  since  I am on a tight schedule.

marceldebout, 2017-09-08:
I just finished implementing a solution for this issue, but will test it on monday before releasing it.

closed 2017-09-11 by MagazinoJenkins

marceldebout, 2017-09-11:
I published the changes and would be happy if you can give me some feedback ;-)

xepost, 2017-09-12:
Hi Marcel ,

Sorry  for the late reply. I have  tested  with 2 Dart and one Ace model all  working  fine. My tests  were simply compiling and verifying the image acquisition. I will let you know if  I can do more tests.