magazino/pylon_camera #35: White Balance disabling?

Crusty82, 2018-03-20:
I connected a Basler acA1920-40uc USB3 camera and set up a profile on the cam in the pylon camera suite utility, among others enabling White Balance.

When running the pylon_camera stack, somehow this setting gets overwritten even if I remove this option from the yaml config file entirely, and the camera drops back into a non-white balanced image.

Is there any way to have the stack not modifying the internal camera params / overriding their settings there with the stack's defaults? Am I using it wrong?

thinkyo, 2018-06-05:
I have the same problem. There is no function for white balancing. Will it be added?

Crusty82, 2018-06-05:
I have found a temporary workaround, where I:

1. Open the camera in pylon_viewer
2. Set all of the settings as I like them.
3. Save the settings as a new profile
4. Set the profile as the startup profile.
5. Disable here in `pylon_camera` to overwrite the settings by uncommenting this line:
https://github.com/magazino/pylon_camera/blob/indigo-devel/include/pylon_camera/internal/impl/pylon_camera_usb.hpp#L71

And then rebuilding. This disables the force-loading of the default user profile on the cam that overrides all of your settings.

matheusns, 2018-06-20:
Hello guys, is there a way to implement this feature to the package, something like a parameter to disable this default profile enabling?

r1d1, 2019-02-27:
The workaround worked for us too but a proper feature would be nice :)
Thanks for the tip !

wayne-li-bc, 2019-07-22:
@thinkyo can you give some details on how you added white balance parameter to the package? We ran into the same params overwriting problem.