magazino/pylon_camera #36: CPU Usage

flajolet, 2018-03-24:
Hello,

  Thanks for this great package!

  I have run some tests to compare the CPU usage of the official pylonViewer and pylon_camera when grabbing images at a constant fps with a GigE camera. The difference is quite significant, see procedure and results below. What is causing such a difference? Could it be because  "the default node operates in Software-Trigger Mode" in pylon_camera?

**Settings** (same settings for pylonViewer and pylon_camera):
  Host computer: Ubuntu 16 kernel v4.4.0-generic
  Network setup: Basler plugged straight into the ethernet port using a POE injector
  NIC of the host computer: Intel I217-LM
  Model: Basler ace GigE
  Image encoding: BayerRG8
  MTU: 9000 (set both on the camera and on the NIC of the host computer)
  FPS: 15
  Inter-packet delay: 0

`top` reports a CPU usage of:
-  ~6% when using pylonViewer (Bandwith Manager > start analysis)
-  ~55% when using pylon_camera

flajolet, 2018-03-26:
I figured it out: [the raw image is rectified even if no one is listening to the rectified topic](https://github.com/magazino/pylon_camera/blob/indigo-devel/src/pylon_camera/pylon_camera_node.cpp#L451-L460) when the camera info is provided.

NikolasE, 2018-03-26:
Great find, you are right, this should be disabled, if the rectified image is never used. 
We need to find a solution that works for the streaming interface (via topics) as well as the action-based interface.

flajolet, 2018-03-26:
`cv_bridge_img_rect_` does not seem to be used at all by the action-based interface so it seems that it would be possible to simply move [this section](https://github.com/magazino/pylon_camera/blob/indigo-devel/src/pylon_camera/pylon_camera_node.cpp#L451-L460) right before publishing the rectified image in the same [if scope](https://github.com/magazino/pylon_camera/blob/indigo-devel/src/pylon_camera/pylon_camera_node.cpp#L433-L436) without impacting anything else.

marceldebout, 2018-03-28:
Thanks! You're right, I'll push your proposed fix soon.

commit 44461219 (2018-03-28): Bugfix for useless rectification

Moved rectification part in the scope of publishing the result.
This resolves #36.
Thanks to @flajolet for your contribution
closed 2018-03-28 by MagazinoJenkins