magazino/pylon_camera #47: Error Payload data on publish images

edmodi, 2018-09-17:
Hi, I newer in pylon_camera node. I'm try to connect a Basler daA1280 USB3.0 in USB2.0. Pylon Viewer and ROS node run ok but when I run rqt_image_view to watch the image, the pylon node show this error message:

`Pylon camera returned invalid image! Skipping`
`Error: 3791651346 Payload data has been discarded. Payload data can be discarded by the camera device if the available bandwidth is insufficient.`
`Error: Grab was not successful`

If I run first rqt_image_view and then pylon_camera_node,  the error does not appear. Why?

Thanks in advanced

GianlucaCerilli, 2018-09-28:
HI, I have the same problem. Did you find the solution @edmodi ?

edmodi, 2018-10-01:
> HI, I have the same problem. Did you find the solution @edmodi ?

Not yet. I'm try but I can not solved it

edmodi, 2018-10-24:
Hi, I change DeviceLinkThroughputLimitMode parameter to Off and run ok

@GianlucaCerilli  could you confirm it?

Thanks

MamounGharbi, 2018-11-06:
Hello!
@edmodi @GianlucaCerilli any news in this side?
(same problem, can't figure out how to change the param in the code)

cheers!

GianlucaCerilli, 2018-11-07:
Hi @MamounGharbi,

Same problem for me..Maybe @edmodi can help us. 

Cheers

megantuttle, 2019-11-15:
I'm sure you have it figured out by now, but in case someone else comes looking, I had the same issue and I also resolved it by setting the DeviceLinkThroughputLimitMode to off.  I added the following to `pylon_camera_usb.hpp` under the `PylonUSBCamera::applyCamSpecificStartupSettings` function:
```
cam_->DeviceLinkThroughputLimitMode.SetValue(Basler_UsbCameraParams::DeviceLinkThroughputLimitMode_off);

hannespimmel, 2020-12-27:
I solved the problem by increasing the Maximum Transfer Size. 
You find this parameter in pylon Viewer at Stream Parameters and with the ros camera driver with the rosservice: "rosservice call /pylon_camera_node/set_max_transfer_size 150000"