magazino/pylon_camera #48: Visual artifacts on image_rect ROS topic when using "bayer_rggb8" encoding

davidkh1, 2018-09-26:
When ROS-Driver for Basler Cameras is configured to use `bayer_bggr8` efficient encoding, visual artifacts appear in rectified images. Image stream on `image_raw` topic is OK in both cases.

What is the root cause of this issue?

Attached:
1) example of a problematic image from `image_rect` topic when in .yaml file
`image_encoding: "bayer_rggb8"` 
2) image is OK when `image_encoding: "rgb8"` in yaml file.

[image]
[image]

schornakj, 2019-04-15:
I've run into this as well. In other CV applications a de-Bayering step is required before rectification to calculate the color values for each pixel from their compressed representation. I think the problem is that at line 521 below the raw image data is passed to `rectifyImage` without accounting for compression, which results in the loss of the compressed colors as the pixels get shifted around.

https://github.com/magazino/pylon_camera/blob/2a7a06803e04ce1658ed423eab63491b5892dc7c/src/pylon_camera/pylon_camera_node.cpp#L513-L523

This might be tricky to fix while still keeping the node's image processing pipeline generic to image formats. One option would be to check the encoding at line 517 and convert to an uncompressed image format if needed.

fho306, 2020-01-17:
I ran into this exact issue as well and found two ways to fix it:

A. Convert the image before rectification:
`cv::cvtColor(cv_img_raw->image, cv_img_debayer_->image, cv::COLOR_BayerBG"BGR);'
'cv_bridge_img_rect_->encoding = "rgb8";'

B. Publish the rectified image via image_transport instead of ros::Publisher*
See this pull request on the newer version of this driver: 
https://github.com/basler/pylon-ros-camera/pull/11
This is the better solution in my opinion, as it is still generic to image formats and I'd want to use image_transport for publishing images anyways.

Because of solution B I actually think the issue is that ros::Publisher* does not support the 'bayer_bggr8' encoding.