magazino/pylon_camera #55: How to set ROI?

ancabilloni, 2019-05-12:
I would like to use ROI to get a smaller focus portion of the image, in hope to has less data to pack and unpack, hence increase performance (lower latency). I see that is an available option on the camera message but how do I set this? And if i use this, would it affect the latency in the negative way since introduce extra process to pylon camera node?

Hope for some input. Thanks!

argolomb, 2019-05-14:
Same issue here 

I'm trying with ex. camera.Width.SetValue(xxx) anda a get the message: 

genicam.AccessException: Node is not writable. : AccessException thrown in node 'Width' while calling 'Width.SetValue()' (file 'IntegerT.h', line 77)

Works when I use AutoFunctionROIWidth node, but for me isn't seen correct.

NikolasE, 2019-07-25:
You can set the ROI with the ~set_roi service:

rosservice call /pylon_camera_node/set_roi "target_roi: {x_offset: 1220, y_offset: 1000, height: 200, width: 400, do_rectify: false}