magazino/pylon_camera #56: unable to process source

ChulhoonJang, 2019-05-21:
Hi,

I am trying to install pylon_camera SDK, but I got the error when I ran 'rosdep update'

> ERROR: unable to process source [https://raw.githubusercontent.com/magazino/pylon_camera/kinetic-devel/rosdep/pylon_sdk.yaml]:

I am using kinetic so I replaced 'indigo' with 'kinetic'. that's it.

What is the problem?

acarrillo, 2019-06-17:
@ChulhoonJang , you probably figured it out by now, but the issue is that there is no kinetic release branch and you have to use `indigo-devel`. If you try the URL you provided yourself, you will see that it `404`s.

It is OK to use `indigo-devel` on Kinetic.

ChulhoonJang, 2019-06-24:
@acarrillo Thank you so much. I did install the indigo-devel and verified the driver was working well. I appreciate it.