magazino/pylon_camera #58: Cameras keep losing connection

yasmeenel3sh, 2019-06-20:
I am trying to synchronize multiple cameras together ,I added ptp support to the package and fetching chunk timestamp  ,also hardware triggering , however when I connect the 2 cameras ,they don't send data with the required frame rate ,then one of them disconnects and when tries to reconnect it gives an error that it is used by another application, Did anyone face this issue before ,please help?

yasmeenel3sh, 2019-06-20:
How does the driver maps the cameras ,is it possible when one camera disconnects the other points to its space ,

NikolasE, 2019-07-25:
How do you start the cameras? Do you configure the device_user_id for each camera?

JeanAlexisBoulet, 2019-10-02:
Hi, we are facing similar issue here with a 5 usb camera setup.
Did you have any success resolving your issue?

yasmeenel3sh, 2019-10-06:
> Hi, we are facing similar issue here with a 5 usb camera setup.
> Did you have any success resolving your issue?

From what I understand is that the mapping of the cameras gets changed I don't know why, so what I did is I increased the delay that the code waits for before reconnecting to the camera so that it has time to show itself again, also I worked around with the interpacket delay and frame transmission delays  so that not all cameras send packets at the same time, however I was using an Ethernet cameras not usb ones, https://github.com/yasmeenel3sh/pylon_camera , this a link to my modified package it has more features including internal timestamping and hardware triggering (though not tested on usb)

yasmeenel3sh, 2019-10-06:
if ( pylon_camera_->isCamRemoved() )
    {
        ROS_ERROR("Pylon camera has been removed, trying to reset");
        delete pylon_camera_;
        pylon_camera_ = nullptr;
        for ( ros::ServiceServer& user_output_srv : set_user_output_srvs_ )
        {
            user_output_srv.shutdown();
        }
        set_user_output_srvs_.clear();
        ros::Duration(1.5).sleep();  // sleep for half a second
        init();
        return;
    }
This part in pylon_camera_node.cpp , i changed the value from 0.5 to 1.5